MotorolaMobilityLLC/kernel-msm #254: Request sources

wh0dat, 2020-05-19:
OPPS27.91-179-9-11
PPBS29.73-81-5-8
QPG30.82-23-1

chenyt9, 2020-05-21:
@wh0dat ，
Could you pls tell us the specific project name, then we will start preparing for the release. Thank you!

chenyt9, 2020-05-21:
Hi， @wh0dat ，
We are preparing to release work, but it will take about two weeks, please be patient, thank you!

wh0dat, 2020-05-21:
Hi and thanks for your reply
I was interested in trying to get Hannah a working 4.9 kernel so i thought
maybe can try to add to surfna's

~Barry Allen

chenyt9, 2020-06-03:
@wh0dat ,
Version released:
PPBS29.73-81-5-8 - surfna
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-PPBS29.73-81-5-8
OPPS27.91-179-9-11 - rhannah
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-OPPS27.91-179-9-11
Thanks.

closed 2020-06-03 by chenyt9