MotorolaMobilityLLC/kernel-msm #256: which defconfig for moto g6 (ali)

nathandiepeveen, 2020-06-08:
Been looking for it for a while now. Though I might aswell ask. Which defconfig should I use for building? There are quite alot and I've been googling for ages and yet no answer.. Thanks alot!

charleseb, 2020-06-08:
Hi @nathandiepeveen,

Could you describe the version you are building, please?

Thanks,
Charles

nathandiepeveen, 2020-06-08:
Hi there! Thanks for your response @charleseb.
I'm building version 'kernel-msm-pie-9.0.0-release-ali'

Thanks again,
Nathan

charleseb, 2020-06-08:
Hi @nathandiepeveen,

Could you refer to a tag, please? Sometimes we use the same branch to more than one tag. Then there is better to get a tag to build.

Thanks,
Charles

nathandiepeveen, 2020-06-08:
Hi! That is the exact name of the tag. But I also found this version for Android 9: 
kernel-msm-MMI-PPS29.55-24.

nathandiepeveen, 2020-06-08:
Hi @charleseb!

Already found something? If not, sorry for the interuption. I'm just very excited. Anyway, thanks for helping.

Nathan

charleseb, 2020-06-09:
Hi @nathandiepeveen,

I need some time to prepare the readme file to this version.
It will be available by Friday.

Thanks,
Charles

nathandiepeveen, 2020-06-09:
Alright @charleseb 

Sounds good to me!

I appriciate it alot. Thanks!

Nathan

nathandiepeveen, 2020-06-15:
Hi @charleseb,

How are you? I was wondering how things are going with the readme file since Friday has passed already. Would love to hear from you,

Nathan

charleseb, 2020-06-15:
Readme file released:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-PPS29.55-24.txt

closed 2020-06-15 by charleseb