MotorolaMobilityLLC/kernel-msm #257: [REQUEST KERNEL]

riansouzasantos, 2020-06-12:
Hello friends, I'm looking for the kernel of two devices:

Pokerp (Moto E6 Plus 2/32)
Pokerp_64 (Moto E6 Plus 4/64)

For study purposes, compilations and maintaining the famous custom rom's. Everything is ready, but I urgently need the kernel source.

chenyt9, 2020-07-03:
Hi，@theczeman , 
Could you tell me which version do you need? Thanks.

closed 2020-09-14 by charleseb

ItsKartik1169, 2025-06-25:
Hi I am new to this building and etc. , can someone build the kernel using source code and provide the download link for that kernel or tell me how to make a custom kernel for my phone, my phone is pokerp_64 (Moto E6 Plus) (4/64) and currently I am using the pixel experience + ( plus ) 12.1 GSI right now. Please support me as soon as possible