MotorolaMobilityLLC/kernel-msm #261: [Request] Moto One Fusion + (liber) kernel sources - QPI30.73-16-5

Hasaber8, 2020-06-25:
Hey,
Could you release kernel sources for this version ?
Thanks

chenyt9, 2020-06-28:
@Hasaber8 ，
We are preparing to release work, but it will take about two weeks, please be patient, thank you!
BRs.
Chenyt9

Hasaber8, 2020-07-04:
> @Hasaber8 ，
> We are preparing to release work, but it will take about two weeks, please be patient, thank you!
> BRs.
> Chenyt9

There is a confusion regarding exactly which chipset this device has, in India its marketed as 730G but in Europe its 730, can you clarify exactly which chipset its having?
-Rohan

Hasaber8, 2020-07-22:
> @Hasaber8 ，
> We are preparing to release work, but it will take about two weeks, please be patient, thank you!
> BRs.
> Chenyt9

Just received an update! It will be nice if you can release QPI30.73-16-5-2.
Rohan

chenyt9, 2020-07-23:
@Hasaber8 ，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPI30.73-16-5
QPI30.73-16-5-2 is in another workflow, we are preparing.
Thanks.

Hasaber8, 2020-07-23:
> https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPI30.73-16-5

Thank you for the release

closed 2020-07-23 by Hasaber8

chenyt9, 2020-08-04:
@Hasaber8 ，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPI30.73-16-5-2
Thanks.