MotorolaMobilityLLC/kernel-msm #269: Request kernel source for sofiap

amarc78, 2020-07-23:
Can you guys please release the kernel source for sofiap QPRS30.80-58-3? Or will the source for sofiar work for sofiap (US retail)?

amarc78, 2020-07-23:
Actually, I see sofiar source wasn't actually released.

chenyt9, 2020-09-15:
Hi， @amarc78 
Here is a version for sofiar：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPE30.79-41
I have started to prepare for the release of this version（sofiap QPRS30.80-58-3）. Due to some processes, it will take about two weeks. Please be patient, thank you!

amarc78, 2020-09-16:
That's ok. It's all good bro. The sofiar kernel source been working fine. Thanks!

chenyt9, 2020-10-15:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPRS30.80-58-3
Thanks.

closed 2020-10-15 by chenyt9