MotorolaMobilityLLC/kernel-msm #274: [Request] Devicetree for 4.19 kernel (kona/lito for the time being)

Vachounet, 2020-08-16:
As reverted here, https://github.com/MotorolaMobilityLLC/kernel-msm/commit/db02045163b7e908b3423fef3cb8215373f21d1d, we're now unable to build dtb/o.

Could you please release them as a separate repo ?

Thanks.

Vachounet, 2020-08-27:
Bump @charleseb @awheeler-mmi

Vachounet, 2020-09-12:
Bump :D

If you need more details, let me know.

awheeler-mmi, 2020-09-14:
Hey @Vachounet,

With the new kernel/GKI refactoring, Qualcomm close-sourced their DT sourcefiles.  So we no longer have a license to share these files with the community.  We will request Qualcomm to open-source their DT source, again.

closed 2020-09-14 by charleseb

konradybcio, 2020-09-14:
> Hey @Vachounet,
> 
> With the new kernel/GKI refactoring, Qualcomm close-sourced their DT sourcefiles. So we no longer have a license to share these files with the community. We will request Qualcomm to open-source their DT source, again.

The files were shared under GPL2-Only. They cannot close-source them "just like that". This violates the software freedom of source availability.

Reference:

https://portland.source.codeaurora.org/quic/la/kernel/msm-4.19/commit/arch/arm64/boot/dts?h=LA.UM.8.12.r1-13500-sm8250.0&id=c85db797af86a29c00f1ccb0e2be89ebbe18896c

awheeler-mmi, 2020-09-14:
hey @konradybcio,

First I can't speak for QC, and I am also not an OSS lawyer.  So the below comments are only based on my observations, as someone looking at the code, and someone that has looked at open-source licenses before.

Let me clarify that I don't precisely know what QC has done.  I just see that the original DTS files are removed, and that new DTS files are now part of our "proprietary" code.  I'm not sure if they are based on the original, open-sourced DTS files, "clean-room" re-implementations, extracted from the .dtb binaries, etc.

The original, GPL'd files are still open-sourced, and you are still free to use and modify those files just as before.  But I believe that QC can "fork" and close-source files that they create -- as long as the "forked" versions contain no community-originated patches.  If the files were community-modified, then they would need to attain an additional license for those files, from the community member(s) that made the changes.  In this case, though, I don't believe that QC has taken any community patches.

Again, I'm not an OSS lawyer, so I may be mistaken.  But this seems to be a regular "dual license" scenario to me.

Vachounet, 2020-09-15:
OnePlus and Xiaomi released them, pretty sure that you should be able to do it too :D

neobuddy89, 2020-09-15:
Xiaomi released without proprietary DTS files. I am sure, Motorola can do
this as well.

Motorola is being quoted as good example on how to release source (not
squashing changes unlike others). This decision of not sharing source will
surely hamper developement ecosystem badly.

awheeler-mmi, 2020-09-15:
Hey @Vachounet, can you point me to the OnePlus and Xiaomi releases?  I see in OnePlus's 8250 kernel where the DTS files are [removed](https://github.com/OnePlusOSS/android_kernel_oneplus_sm8250/tree/oneplus/SM8250_Q_10.0/arch/arm64/boot/dts/qcom), but not where they have moved them to.  QCOM's release to us includes these DTS files under vendor/qcom/proprietary/devicetree-4.19.  (QC releases their closed-source Android code to us in this "proprietary/" subdirectory.)

> Motorola is being quoted as good example on how to release source (not
> squashing changes unlike others). This decision of not sharing source will
> surely hamper developement ecosystem badly.

@neobuddy89  This is about files that QCOM releases to us under their "proprietary" license, not via open-source.  We don't have a license to share files from their "proprietary" releases.  I will ask QC to open-source this content, or allow us to open-source them.  It will help our case if you can point out where other vendors have already released these files.

Thanks,

-andrew

awheeler-mmi, 2020-09-15:
I just noticed [this thread](https://www.reddit.com/r/Android/comments/iiybi5/android_aosp_can_now_boot_off_mainline_linux/g3c39fw/?context=3), where a OnePlus dev is running into the same issue.   Though they mention that OnePlus is still releasing it, albeit an older version.  If OnePlus is releasing it then I'll make sure we can, too.

Hlcpereira, 2020-09-28:
@awheeler-mmi both oneplus and Xiaomi are releasing. Oneplus is updating the dts too afaik

Hasaber8, 2020-10-26:
> I just noticed [this thread](https://www.reddit.com/r/Android/comments/iiybi5/android_aosp_can_now_boot_off_mainline_linux/g3c39fw/?context=3), where a OnePlus dev is running into the same issue. [...]

Any update on this? @awheeler-mmi

awheeler-mmi, 2020-10-26:
I opened a support issue with QCOM, and asked them to open-source these files.  I'll work on escalating the request.

Hasaber8, 2020-10-26:
> I opened a support issue with QCOM, and asked them to open-source these files. I'll work on escalating the request.

Thank you for the prompt response.

Hasaber8, 2020-11-02:
> I opened a support issue with QCOM, and asked them to open-source these files. I'll work on escalating the request.

First of all I want to thank you for being so patient and answering our questions, but I still dont understand how Xiaomi and OnePlus have releases the dts for 4.19 kernel but Motorola hasn't. This is really causing a huge inconvenience on our part as developers in booting custom kernel and android builds. I know @awheeler-mmi you are trying you best to talk with QCOM and get us those dts, but please look into this issue even more , because as I said before other OEMs have released dts albeit Xiaomi ones are broken.

awheeler-mmi, 2020-11-02:
Hey @Hasaber8,

I have heard back from QCOM, where they confirmed that these files are proprietary.   So I have escalated this issue to our business team.

W/rt Xiaomi and and OnePlus, we can't release them just because they did.  I did raise this specific issue w/ QCOM, but they didn't bite.  This is going to take some time.

Hasaber8, 2020-11-02:
> Hey @Hasaber8,
>
> I have heard back from QCOM, where they confirmed that these files are proprietary. So I have escalated this issue to our business team.
>
> W/rt Xiaomi and and OnePlus, we can't [...]

This is really sad, but still thank you for putting our word in front of the respective teams. Please let us know when something big happens.

Hlcpereira, 2020-11-02:
@awheeler-mmi I'm sorry if I'm saying something completely irrational/dumb, but as the kernel is under a GPL-2.0 license wouldn't that mean that once anything included in the kernel, despite the license that the driver/dts would have, could be made public without any issues exaclty because they are under the LK GPL-2.0 license then, therefore nullifying the license of confidental and proprietary code of those?

Hasaber8, 2020-12-22:
> Hey @Hasaber8,
>
> I have heard back from QCOM, where they confirmed that these files are proprietary. So I have escalated this issue to our business team.
>
> W/rt Xiaomi and and OnePlus, we can't [...]

Hey @awheeler-mmi ,
Any updates from the business team?

awheeler-mmi, 2020-12-23:
No update.  I'll escalate, once again.

xboxfanj, 2021-01-20:
For reference, Google, Xiaomi, OnePlus, ZTE, Asus, Samsung, and Oppo released 4.19 DTS. It may be a different question if the CAF unmodified DTS is proprietary and if OEMs can release theirs.

MartinX3, 2021-01-20:
And Sony
https://github.com/sonyxperiadev/kernel/commit/fb370d8213a692cf8ed307267f5ae9ec86fc07df

awheeler-mmi, 2021-01-20:
Thanks for the Sony link. This will help.

Hasaber8, 2021-01-20:
> Thanks for the Sony link. This will help.

Hopefully it will good news from your side soon.

awheeler-mmi, 2021-01-20:
Yeah, I'm trying again.  I just shot an email to my closest contact at QCOM that's in charge of these sorts of things.  Hopefully I get some traction.  I don't think anyone wants to keep this stuff from developers, it's just hard to get stuff like this fixed.  Thanks for the patience -- and continuing to provide "competitor" info, which definitely helps.

awheeler-mmi, 2021-01-21:
Good news, QC gave us approval to share the DTS files.  We just need to figure out how to filter out the non-relevant dts files (QC's only stipulation), and come up with an efficient release mechanism.

Sadly, our main open-source engineer, Charles, is no longer with our company.  (He's off to a new adventure in Europe - wish him well!)  So give us a bit of time to work this out.

xboxfanj, 2021-01-21:
Thanks Andrew, much appreciated!

Vachounet, 2021-01-21:
> Good news, QC gave us approval to share the DTS files. We just need to figure out how to filter out the non-relevant dts files (QC's only stipulation), and come up with an efficient release [...]

Awesome, thank you :)

Hasaber8, 2021-01-21:
> Good news, QC gave us approval to share the DTS files. We just need to figure out how to filter out the non-relevant dts files (QC's only stipulation), and come up with an efficient release [...]

Hope Andrew does amazing at his new adventure!!
And thank you for the amazing news!

raza231198, 2021-01-21:
> Good news, QC gave us approval to share the DTS files. We just need to figure out how to filter out the non-relevant dts files (QC's only stipulation), and come up with an efficient release [...]

Nice, thanks 😬

Hasaber8, 2021-02-20:
> Good news, QC gave us approval to share the DTS files. We just need to figure out how to filter out the non-relevant dts files (QC's only stipulation), and come up with an efficient release [...]

Still waiting :-\

raza231198, 2021-03-13:
@awheeler-mmi any update sir?

seisdr, 2022-05-04:
So i want to increase the virtual address space، is that dtb required ? because the kernel refuse to boot with 48vas 4k page

MartinX3, 2022-05-04:
> So i want to increase the virtual address space، is that dtb required ? because the kernel refuse to boot with 48vas 4k page

This has nothing to do with this issue ticket.
Please don't spam.