MotorolaMobilityLLC/kernel-msm #276: Moto G5S (XT1799-2) kernel source request - NZS26.68-117-10

JarlPenguin, 2020-08-19:
Can you please release the kernel sources for this tag? Thank you.

xiongkg1, 2020-08-28:
> Can you please release the kernel sources for this tag? Thank you.

Dear JarlPenguin, Can you confirm the version information again? 
The steps are as follows: Settings ---> About ----> ZUI Vsersion
Here is a version release of montana _retail:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPP26.102-41
montana _retail and montana _retcn the code is basically the same. Please see if it will help you.

JarlPenguin, 2020-08-29:
Hi,
The ZUI version is 3.1.86-128 ST. With the provided kernel source unfortunately front camera does not work.

xiongkg1, 2020-09-01:
> Hi,
> The ZUI version is 3.1.86-128 ST. With the provided kernel source unfortunately front camera does not work.

Dear JarlPenguin
      Nzs26.68-117-10 and 3.1.86-128 st are not officially released by us. How do you get this version? Is your mobile phone purchased in the market or customized through channels?

JarlPenguin, 2020-09-02:
I infact do not have the device itself, but I know some people have it. If the kernel source is the same for both variants, I will try to look for a solution. Thank you for your help.

closed 2020-09-02 by JarlPenguin