MotorolaMobilityLLC/kernel-msm #277: [Request] Motorola One Zoom - QPH30.29-Q3-28-13

betaxab, 2020-08-24:
Hi,

Could you release kernel sources for this version?

Thanks.

charleseb, 2020-08-27:
Hi @betaxab,

We are starting the process to release this version. It will be available by Tuesday.

Thanks,
Charles

betaxab, 2020-08-28:
@charleseb hi, could you upload this kernel's build readme? Thanks!

charleseb, 2020-09-10:
Hi @betaxab,

I'm working to release the readme file. It will be available by Monday.

Thanks,
Charles

betaxab, 2020-09-13:
hi, @charleseb ,

Could you release Motorola E7 ginNA's readme?

Thanks! 👍

alexenferman, 2020-11-20:
Please release the Readme for Moto E 2020 Ginna!

chenyt9, 2020-11-23:
@betaxab @alexenferman ,
For Motorola E7 ginNA's readme:
Pls refer to :https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-QPG30.82-95.txt
Thanks.

closed 2020-12-22 by betaxab