MotorolaMobilityLLC/kernel-msm #280: [Request] z4 kernel for QPFS30.130-15-11-5 and Device Tree

Swagman89, 2020-09-03:
Please release the Z4 Android Q kernel source and if possible a device tree, building TWRP for the z4 to bring motorola back to where it was in the phone development world.

Swagman89, 2020-09-03:
Added clarification, I need the Unlocked Variant for the USA.

charleseb, 2020-09-10:
Hi @Swagman89,

We are starting the process to release this version. It will be available by next week.

Thanks,
Charles

awheeler-mmi, 2020-09-14:
@Swagman89, by "device tree" do you mean the device's build configuration makefiles (e.g. messi.mk, etc.), or the DTB source?

Swagman89, 2020-09-14:
@awheeler-mmi I am intending to use this to build twrp for the device. I'm still a little green on all the exacts. I was hoping to just get what's referred to in other documentation as the device tree that's used when you compile TWRP or truthfully any Android build. I wasn't sure if you'd provide it or not. If you're not sure still based on my explanation then just let me get anything and everything. I'll sort through it.

awheeler-mmi, 2020-09-14:
Ok, based on what I found [here](https://gerrit.twrp.me/admin/repos), the TWRP project is referring to the build configuration.  This isn't something we can open-source.  Our product configuration is pretty huge, convoluted, and involves some closed-source code, too.

Swagman89, 2020-09-14:
Kinda figured as much. But my favorite saying is closed mouths never get
fed. Release what you can and I'll take it from there! Thanks.

Swagman89, 2020-09-17:
@awheeler-mmi I'll take anything you can give me.

charleseb, 2020-09-24:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPFS30.130-15-11-5

closed 2020-09-24 by charleseb