MotorolaMobilityLLC/kernel-msm #281: With mobile data disabled, it consumes mobile data and can surf the internet on moto g7 plus

JulyMorga, 2020-09-07:
After updating from Android 9 to Android 10 on the moto g7 plus. With the mobile data deactivated, and exit the airplane mode, or lose the Wi-Fi signal, or turn on the cell phone. Mobile data is consumed, generating a charge to the user. This error is frequently manifested.
The error occurs from the first version of Android 10 (build QPW30.61-21), and the last one also (build QPW30.61-21)

This error occurs on all moto g7 plus

JulyMorga, 2020-09-10:
Hello, Do you need any other information?

The error looks like a failure to initialize the gprs module. Once the error occurs, you can turn the wifi off and on and the error will continue to occur. When putting the airplane mode and taking it out, the error sometimes stops happening. But when you turn off the cell phone or activate the airplane mode again, and turn on or off the airplane mode, the error can occur again.

I confirm that the error does not occur on Android 9 and started to occur when updating to Android 10.

The error happens in version release (build QPWS30.61-21-14)

Thank you very much

awheeler-mmi, 2020-09-10:
Hey @JulyMorga,

This is, indeed, a bad issue that you're experiencing. Could you move your message to the [g7 support forum](https://forums.lenovo.com/t5/moto-g7-play-moto-g7-moto-g7-power-moto-g7-plus/bd-p/moto_g7_play_power_plus)?  Our support forums have staff (and customers) that can help with this sort of thing.

Thanks!

-andrew

JulyMorga, 2020-09-11:
Hellow @awheeler-mmi , thanks for answering!
 
Yes, I already put it on one occasion. I also took it to official technical service 5 times, they denied me the problem. They changed my cell phone and the same thing happens when updating to Android 10.

It is not a sim card problem, or applications.

Just do it again in the forum, with more description:
https://forums.lenovo.com/t5/moto-g7-play-moto-g7-moto-g7-power-moto-g7-plus/MOTO-G7-PLUS-With-mobile-data-disabled-it-consumes-mobile-data-and-can-surf-the-internet/m-p/5038225

awheeler-mmi, 2020-12-22:
Closing this issue, since we have the Motorola support thread about it.

closed 2020-12-22 by awheeler-mmi