MotorolaMobilityLLC/kernel-msm #282: Potter OPS28.85-17-6-2 - DTB missing for P3B?

thomasmey, 2020-09-08:
Hi,

what is the correct dtb file for "product/variant": "potter XT1685 32GB P3B"?
the source tree contains only these

arch/arm/boot/dts/qcom/msm8953-potter-p1b.dtb
arch/arm/boot/dts/qcom/msm8953-potter-p0a.dtb
arch/arm/boot/dts/qcom/msm8953-potter-p1a.dtb
arch/arm/boot/dts/qcom/msm8953-potter-p2b.dtb
arch/arm/boot/dts/qcom/msm8953-potter-p3a.dtb
arch/arm/boot/dts/qcom/msm8953-potter-p2a2.dtb
arch/arm/boot/dts/qcom/msm8953-potter-p2a.dtb

no "P3B like above string indicates

awheeler-mmi, 2020-09-14:
Hey @thomasmey, we don't have this code easily searchable, but if there is no p3b.dtb then it is likely p3b.dtb.  The difference between hardware rev 3a and 3b was likely very small, and didn't impact the dtb.

thomasmey, 2020-09-14:
Hi,

Thanks for the update. In the mean time I did find out that p3a seems to work correctly also.

Sorry for the fuzz.

closed 2020-09-14 by charleseb