MotorolaMobilityLLC/kernel-msm #285: Moto G9 ( india )

vvknd, 2020-09-13:
Recently purchased moto g9 ( india ) having Snapdragon 662 according to website but in about phone baseband version is shown as  M4250_XX.XXX.XX.XXR which is for Snapdragon 460.

For Snapdragon 662  baseband version should be  M6115_XX.XXX.XX.XXR.

charleseb, 2020-09-14:
Hi @vvknd,

Could you describe the version tag(s) you need, please?

Thanks,
Charles

vvknd, 2020-09-15:
(. Issue )

Here is the screenshot.
[image]

Why SD 662 phne showing baseband of sd460 ?
is Motorola cheating ?

badmaan, 2020-09-16:
Those chipsets has a unified BSP from Qualcomm. This is not related to Motorola.

PythonLimited, 2020-09-16:
> Those chipsets has a unified BSP from Qualcomm. This is not related to Motorola.

I looked up the modem versions at qcom, but all sdm662 have M6115 or SDM6XX.GEN.PACK.1 as unified modem version.
In fact the sdm460 isnt even included in the sdm662 BSP.

badmaan, 2020-09-16:
No.
@[image]

PythonLimited, 2020-09-16:
> No.
> [image]

I'm talking about the BSP.
Not the design guidelines or docs.
Send the about.html that includes both.

badmaan, 2020-09-16:
> > No.
> > [image]
> 
> I'm talking about the BSP.
> Not the design guidelines or docs.
> Send the about.html that includes both.

[image]

PythonLimited, 2020-09-16:
> > > No.
> > > [image]
> > 
> > 
> > I'm talking about the BSP.
> > Not the design guidelines or docs.
> > Send the about.html that includes both.
> 
> [image]

What is that repo name?
I have them separated and only for R...
[image]

badmaan, 2020-09-16:
> > > > No.
> > > > [image]
> > > 
> > > 
> > > I'm talking about the BSP.
> > > Not the design guidelines or docs.
> > > Send the about.html that includes both.
> > 
> > 
> > [image]
> 
> What is that repo name?
> I have them separated and only for R...
> [image]

snapdragon-mid-2020-spf-1-0_hlos_dev revision 00018.

chenyt9, 2020-09-25:
Hi,@vvknd ,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPX30.30-Q3-38-20
Thanks

vvknd, 2020-09-29:
> Hi,@vvknd ,
> Version released:
> https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPX30.30-Q3-38-20
> Thanks

thanks , but my concern is for baseband version.

vvknd, 2020-09-29:
@chenyt9 my concern is for baseband version.
and thank for code.

awheeler-mmi, 2020-09-29:
Hey @vvknd,

SM4250 and SM6115 both have a Snapdragon X11 LTE modem, and share the same modem binary.  The version number, M4250_XX.XXX.XX.XXR, is our own versioning scheme, and we tend to put the chipset number of the first chipset that the release supports, rather than all of the chipsets that the release supports.

vvknd, 2020-09-29:
@awheeler-mmi please change string to M6115 to rmeove confusion.

thank you.

awheeler-mmi, 2020-09-29:
@vvknd Oh if it were only that easy!

closed 2020-09-30 by vvknd