MotorolaMobilityLLC/kernel-msm #288: [Request]: Odessa QPI30.28-Q3-28-26 Kernel Sources

kytoaq, 2020-09-24:
<h1> Hi Moto!  👋 </h1>
<h3> Please, could you release kernel sources for this version? </h3>
<h2> Thanks 🤨 <h2>

chenyt9, 2020-10-09:
Hi，@kytoaq 
We have released multiple versions starting with QPA30.19-Q3-32. Could you describe the version tag(s) you need, please?
Thanks.

kytoaq, 2020-10-11:
Well, could you provide the source code:

**ODESSA**     **BRLA_CUST**       **QPA30.19-Q3-32-21**
**ODESSA**     **BRLA_CUST**       **QPA30.19-Q3-32-39**

Thank you

chenyt9, 2020-10-13:
Hi,@kytoaq ,
I am starting to trigger the release process, which will take about two weeks. Please wait patiently, Thank you!
Chenyt9

Hasaber8, 2020-11-13:
> Hi,@kytoaq ,
> I am starting to trigger the release process, which will take about two weeks. Please wait patiently, Thank you!
> Chenyt9

Hey @chenyt9 any updates on this?

chenyt9, 2020-11-18:
@kytoaq , 
There is a problem in process.
Could you pls help reconfirm the version number？I only found this version：QPA30.19-Q3-32-39-1. Thanks.

kytoaq, 2020-11-18:
@chenyt9 Of course, I need that version. In the same way I need you to please provide the README file to generate the correct kernel defconfig, Thanks.

chenyt9, 2020-11-18:
@kytoaq ，
What I need to confirm is, we only have a release record for QPA30.19-Q3-32-39-1，I will release QPA30.19-Q3-32-39-1 today. Will it meet your needs?

Hasaber8, 2020-11-18:
> @kytoaq ，
> What I need to confirm is, we only have a release record for QPA30.19-Q3-32-39-1，I will release QPA30.19-Q3-32-39-1 today. Will it meet your needs?

Definitely works!

kytoaq, 2020-11-18:
Yes, I need this release QPA30.19-Q3-32-39-1 as you mentioned. I also request that you provide the README file to generate the correct kernel configuration (defconfig) please, thanks

chenyt9, 2020-11-18:
Okay, no problem, I am preparing to push. The reame file will also be provided, please be patient. Thank you!

kytoaq, 2020-11-18:
Thanks for the support 👍

chenyt9, 2020-11-19:
@kytoaq @Hasaber8 ，
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPA30.19-Q3-32-21
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPA30.19-Q3-32-39-1
Thanks.

closed 2020-11-19 by chenyt9