MotorolaMobilityLLC/kernel-msm #292: [Request] Full techpack drivers for 4.19 kernel (kona/lito)

erfanoabdi, 2020-11-03:
Please release source code of techpack drivers for 4.19 devices
Example device tags: QCS30.58-Q3-44-9-11-2 or QPD30.70-58
Techpack includes: 

- camera
- dataipa
- display
- video
- audio

audio already available at https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-audio-kernel but other drivers missing

Thanks
cc @charleseb @awheeler-mmi @chenyt9

erfanoabdi, 2020-11-10:
@charleseb any updates on this?
we need moto changes on below qcom repos:
https://source.codeaurora.org/quic/la/platform/vendor/opensource/camera-kernel/
https://source.codeaurora.org/quic/la/platform/vendor/opensource/display-drivers/
https://source.codeaurora.org/quic/la/platform/vendor/opensource/video-driver/
https://source.codeaurora.org/quic/la/platform/vendor/qcom-opensource/data-kernel/

(All repos licensed GPL-2.0)

Moto edge (racer) have glitchy screen on source built kernel using display driver from CAF: 
[image]

erfanoabdi, 2020-11-24:
Bump @charleseb @awheeler-mmi @chenyt9

betaxab, 2020-12-22:
hi @charleseb @awheeler-mmi @chenyt9 any updates???

awheeler-mmi, 2020-12-22:
Hey @betaxab. Yes, sorry for delay.  We are working on the release of the techpack drivers.  We should be getting them out shortly - hopefully an early Xmas present.

betaxab, 2020-12-22:
oh, good luck! @awheeler-mmi

charleseb, 2020-12-22:
Hi @betaxab and @erfanoabdi,

Please find the techpack drivers listed below:
QPD30.70-58:
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-video/releases/tag/MMI-QPD30.70-58
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-camera/releases/tag/MMI-QPD30.70-58
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-data-kernel/releases/tag/MMI-QPD30.70-58
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-display/releases/tag/MMI-QPD30.70-58

QCS30.58-Q3-44-9-11-2:
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-data-kernel/releases/tag/MMI-QCS30.58-Q3-44-9-11-2
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-video/releases/tag/MMI-QCS30.58-Q3-44-9-11-2
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-display/releases/tag/MMI-QCS30.58-Q3-44-9-11-2
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-camera/releases/tag/MMI-QCS30.58-Q3-44-9-11-2

Sorry about the delay.

Thanks,
Charles

closed 2020-12-22 by charleseb