MotorolaMobilityLLC/kernel-msm #296: [GPL] Release the Motorola Evolve kernel source

PythonLimited, 2020-11-25:
As stated above, the kernel source for the newly launched Motorola Evolve hasn't been uploaded.
Please do so!

Hasaber8, 2020-12-15:
> As stated above, the kernel source for the newly launched Motorola Evolve hasn't been uploaded.
> Please do so!

That device isn't handled by motorola mobility.

mbissaromoto, 2020-12-15:
Hi @PythonLimited  
Motorola Evolve is handled by Motorola Solutions, not Motorola Mobility. Please, see here: https://github.com/MotorolaSolutions

PythonLimited, 2020-12-16:
> Hi @PythonLimited
> Motorola Evolve is handled by Motorola Solutions, not Motorola Mobility. Please, see here: https://github.com/MotorolaSolutions

I don't see any kernel repos there either.

Can you suggest me a repo where I could open the issue?

mbissaromoto, 2020-12-22:
> > Hi @PythonLimited
> > Motorola Evolve is handled by Motorola Solutions, not Motorola Mobility. Please, see here: https://github.com/MotorolaSolutions
> 
> I don't see any kernel repos there either.
> 
> Can you suggest me a repo where I could open the issue?

Hi @PythonLimited 
Sorry, unfortunately I can't help you with this. Motorola Solutions is a different company, and we have no internal contacts to talk about it. You can maybe try to find a project that has more issues (even if not related to kernel) and open a new one.

awheeler-mmi, 2020-12-22:
Closing this issue, since the OS request is for a Motorola Solutions product.

closed 2020-12-22 by awheeler-mmi