MotorolaMobilityLLC/kernel-msm #298: Motorola Edge (racer) - Kernel source for tag QPD30.114-80

Jleeblanch, 2020-11-29:
@charleseb @awheeler-mmi

Can you please release the kernel source and other relevant sources for tag **QPD30.114-80**?

Thank you,
Joshua

chenyt9, 2020-12-02:
Hi，@Jleeblanch ，We are preparing to release work, but it will take about two weeks, please be patient, thank you!

chenyt9, 2020-12-11:
@Jleeblanch ,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPD30.114-80
Thanks.

erfanoabdi, 2020-12-11:
@chenyt9 this release is missing techpack drivers (display, camera, ..)
please check https://github.com/MotorolaMobilityLLC/kernel-msm/issues/292

erfanoabdi, 2020-12-21:
Please push this reversion for the following repos:

https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-camera
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-display
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-video
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-data-kernel

charleseb, 2020-12-22:
Hi @erfanoabdi,

We are working to release the techpack drivers.

Thanks,
Charles

Jleeblanch, 2020-12-22:
@charleseb We thank you!

mbissaromoto, 2020-12-23:
Hi @erfanoabdi 

Techpack drivers available at:
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-data-kernel/releases/tag/MMI-QPD30.114-80
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-video/releases/tag/MMI-QPD30.114-80
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-display/releases/tag/MMI-QPD30.114-80
https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-camera/releases/tag/MMI-QPD30.114-80

closed 2020-12-23 by mbissaromoto

erfanoabdi, 2020-12-25:
Thanks 
sorry i forgot to ask in previous comment
please push racer tag for "motorola-kernel-modules" repo

mbissaromoto, 2020-12-28:
Hi @erfanoabdi 

It is done. Check here: https://github.com/MotorolaMobilityLLC/motorola-kernel-modules/releases/tag/MMI-QPD30.114-80

Att,
Marcelo