MotorolaMobilityLLC/kernel-msm #303: kernel-msm Vs individual device kernel sources

arunkumar413, 2020-12-11:
Hi Team,

The about section says the repo contains the kernal sources for msm based devices. However, I see that kernel sources for individual devices are being requested by the users. Could you please provide more clarity on the devices  (trade names)  this repo is for. Also how individual kernel sources are different from this repo.

Thanks,
Arun

awheeler-mmi, 2020-12-22:
Hey @arunkumar413, this repo, kernel-msm, is for all of our qcom-based products.  We have corresponding projects for slsi and mtk  Since most of our products are QCOM-based, most of the kernel releases are in this project. 

The "decoder ring" between trade names and our releases is quite complex, due to the number of products, trade names, and releases that we make.  So we tag the release with the build ID, which should be visible on your device, in Settings.

awheeler-mmi, 2020-12-23:
Closing for now.  Re-respond if there are still questions. Thanks!

closed 2020-12-23 by awheeler-mmi