MotorolaMobilityLLC/kernel-msm #306: [request kernel] moto e6 play

GKart, 2020-12-20:
hello and thanks for reading.
I need the kernel of the moto e6 play bali XT2029-1 2/32 to be able to make studies of the operation of the system and the creation of the well-known custom ROMS.

mbissaromoto, 2020-12-21:
Hi @GKart 
For exactly which tag do you need it? Example: POAS29.475-62-26-7

GKart, 2020-12-21:
Thank you for reading.

I need POA29.550.132 please.
by the way, being the legal owner of the device I want to know what is running in the background and for that I need to be told, if possible, how to unlock the bootloader of this device.

GKart, 2020-12-22:
Hi @awheeler-mmi 
thanks
The device is totally solid and although it does not help me, your work on this device is incredible. I cannot find a way to unlock the bootloader since it detects it as unknown to all commands. It has the most limited fastboot menu I've ever seen: fastboot devices and fastboot reboot, nothing more. I think that to develop next time I will buy one with snapdragon.

awheeler-mmi, 2020-12-22:
Yeah, I believe that MTK products use a different flashing tool than fastboot. I have never actually used it, since my team deals more with the QCOM (and SLSI)-based products.

GKart, 2020-12-22:
> Yeah, I believe that MTK products use a different flashing tool than fastboot. I have never actually used it, since my team deals more with the QCOM (and SLSI)-based products.
Thranks

GKart, 2020-12-23:
So there is no solution for the bootloader?

awheeler-mmi, 2020-12-23:
Yeah, there is no supported method for unlocking the Bali bootloader.

GKart, 2020-12-23:
@awheeler-mmi thranks

kytoaq, 2021-01-06:
Hi @awheeler-mmi any update on this? Release would be greatly appreciated, please provide ETA. Thanks.

awheeler-mmi, 2021-01-06:
I'm moving this to [kernel-mtk](https://github.com/MotorolaMobilityLLC/kernel-mtk/issues/39) .

closed 2021-01-06 by awheeler-mmi