MotorolaMobilityLLC/kernel-msm #315: [Request] Moto G 5G (kiev) - QZK30.Q4-40-32 Kernel Sources

GarfieldFernando, 2021-01-16:
Hello!
Can you release kernel sources for this version? Please, thx. :DD

kytoaq, 2021-01-16:
If possible also release dtsi for kernel 4.19 please, being broad importance the device tree of kiev and drivers. Thanks.

Deivid21, 2021-01-18:
Can release the kernel from QZK30.Q4-40-52 and full techpack, please?
Thx!

GarfieldFernando, 2021-01-18:
sorry, release for techpack too, please

mbissaromoto, 2021-01-18:
Hi @GarfieldFernando , @Deivid21  and @kytoaq 

For a better tracking, can you please open a new issue to request this QZK30.Q4-40-52 tag?

As 40-32 is the first Kiev tag we gonna release, our internal process will take some time. After it is done and first tag is released, all next tags can be released right after. 

Att,
Marcelo

GarfieldFernando, 2021-01-18:
Ok, i will do

mbissaromoto, 2021-01-18:
Reopening. This issue will be used to track the 40-32.

alexguna1, 2021-03-11:
> Hi @GarfieldFernando , @Deivid21 and @kytoaq
>
> For a better tracking, can you please open a new issue to request this QZK30.Q4-40-52 tag?
>
> As 40-32 is the first Kiev tag we gonna release, our [...]

Could you pls update, its been more then 2months now

GarfieldFernando, 2021-03-13:
Oh, yeah i will update

alexguna1, 2021-03-13:
> Hi @GarfieldFernando , @Deivid21 and @kytoaq
>
> For a better tracking, can you please open a new issue to request this QZK30.Q4-40-52 tag?
>
> As 40-32 is the first Kiev tag we gonna release, our [...]

The other vendor has already Released, the Kernel Sources for the same chipset. What is the issue with Motorola.

https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/gauguin-q-oss

mbissaromoto, 2021-03-16:
Hi guys,
 
Really sorry for the delay to reply here. Since Charles has left the company, we are in lack of human resource to take care of OSS publications. And as it is the first tag of a product, it takes even more time. We are working in the automation of this process so it doesn't require too much human effort, but it will take time. One more time: Sorry for this and please bear with us :) 

Att,
Marcelo

Hasaber8, 2021-03-24:
Close this issue, as everything needed has been released

mbissaromoto, 2021-04-08:
Released: https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-QZK30.Q4-40-32

closed 2021-04-08 by mbissaromoto