MotorolaMobilityLLC/kernel-msm #322: [Request] Moto G9 Play (Guamp) - QPXS30.30-Q3-38-42-4 Kernel Sources

Deivid21, 2021-01-31:
Hi, can you release kernel sources and full techpack for this version, Thanks!

chenyt9, 2021-02-01:
Hi, @Deivid21 ,
We are starting the process to release this version. We will release release techpack.
It will need about two week. Please wait patiently, Thank you!

kytoaq, 2021-02-16:
Any news about this? Guam has been released but I'm not sure if they are compatible. The kernel is based on LA.UM.8.15.r2-02400-KAMORTA.0 (Bengal). Please when they launch it could you specify the differences between guam (Moto E7 Plus) and guamp (Moto G9/G9 Play) please. If you can include the dtsi and dts from kernel 4.19 that would be great. Thank you.

chenyt9, 2021-02-19:
Version release：https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPXS30.30-Q3-38-42-4.
Thanks.

closed 2021-02-27 by Deivid21