MotorolaMobilityLLC/kernel-msm #323: [Request] moto g pro (sofiap_sprout) - RPR31.Q4U-20-26-2

Vachounet, 2021-02-06:
Hi,

Could you realease kernel sources from the R update please ?

Vachounet, 2021-02-20:
Bump @chenyt9 @mbissaromoto :)

kytoaq, 2021-04-16:
@mbissaromoto @chenyt9 @ccfries release when? Two months have passed and there has been no reply, please indicate when you could do it. Without the urge to interrupt, thank you

kytoaq, 2021-04-16:
If they could upload a newer tag like RPRS31.Q4U-20-30 it would be better. Please include everything you need like drivers and techpack

kytoaq, 2021-05-29:
@chenyt9 any update?? 🤔

kytoaq, 2021-07-16:
Release when? @mbissaromoto @chenyt9 @awheeler-mmi

mbissaromoto, 2022-03-05:
Hi all,

It is released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RPR31.Q4U-20-26-2

Regards,
Marcelo

closed 2022-03-05 by mbissaromoto