MotorolaMobilityLLC/kernel-msm #328: [Reques] Kernel Source from Motorola Moto E7 Plus (guam)

unc0ver-debug, 2021-03-08:
I need the Motorola Moto E7 Plus kernel source (guam).

RainbowCookie32, 2021-03-13:
There's #319 already

https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPZS30.30-Q3-38-42-3

durasame, 2021-03-13:
> There's #319 already
> 
> https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPZS30.30-Q3-38-42-3

Old version, I guess

chenyt9, 2021-05-06:
@unc0ver-debug ，Can you specify the version you need? Thanks.

chenyt9, 2022-09-20:
Hi，@riansouzasantos 
Could you specify which version do you want?
The current version we have released is here:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPZS30.30-Q3-38-42-3
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPZS30.30-Q3-38-66-2

chenyt9, 2022-09-22:
Hi，@all，
Close CR first, if there are still version requirements, please re-open this issue and specify which version you need.
Thanks.

closed 2022-09-22 by chenyt9