MotorolaMobilityLLC/kernel-msm #332: When are you releasing Moto G 5G aka Moto One 5G Ace (kiev) Kernel Source

pandayashas-github, 2021-03-13:
Release the kernel source for Moto G 5G aka Moto One 5G Ace (kiev). It's high time.

Hasaber8, 2021-03-14:
https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-QZK30.Q4-40-62
already released , please close the issue.

Hasaber8, 2021-03-17:
Still waiting on those Device trees for 4.19 @awheeler-mmi

awheeler-mmi, 2021-03-17:
Sorry for the delay.  We are trying to catch up since our main OSS engineer, Charles is no longer on the team.  As an initial experiment with pushing DTS files, I just pushed up **MMI-QZK30.Q4-40-62* in these new gits: 

    - kernel-devicetree
    - kernel-camera-devicetree
    - kernel-display-devicetree

I believe these are all our DTS files.  Let me know if we are missing any.  

Our plan is to add the automation to our regular open-source releases, but it will take us some time to get the kinks worked out.

Jleeblanch, 2021-03-17:
@awheeler-mmi 
You are the man! Thank you xD

Hasaber8, 2021-03-18:
> Sorry for the delay. We are trying to catch up since our main OSS engineer, Charles is no longer on the team. As an initial experiment with pushing DTS files, I just pushed up *_MMI-QZK30.Q4-40-62_ [...]

Thank you very much @awheeler-mmi

mbissaromoto, 2021-04-09:
Closing the issue. If those 3 repos listed by Andrew are not enough, please reopen this issue.

closed 2021-04-09 by mbissaromoto