MotorolaMobilityLLC/kernel-msm #335: [Request] Motorola Edge S Device Tree and Kernel Sources (RRN31.Q1-20-31-3)

electimon, 2021-03-19:
Hello!, Can you please release kernel sources (kernel, techpacks, and devicetree) for nio RRN31.Q1-20-31-3? Thank you 😊

betaxab, 2021-03-19:
Please please @chenyt9

Hasaber8, 2021-03-24:
Bump @awheeler-mmi

Hasaber8, 2021-04-16:
bump @mbissaromoto

betaxab, 2021-05-02:
Requst Motorola Edge S Device Tree and Kernel Sources (RRN31.Q1-20-52-3)

@chenyt9 

I really need PWM related kernel updates, thank you

thanks.

chenyt9, 2021-05-06:
Sorry for delay. Which version?RRN31.Q1-20-52-3 or RRN31.Q1-20-31-3？

chenyt9, 2021-05-06:
Hi，@betaxab ，Which version?RRN31.Q1-20-52-3 or RRN31.Q1-20-31-3？

Hasaber8, 2021-05-06:
RRN31.Q1-20-52-3

chenyt9, 2021-05-06:
@Hasaber8 ，thank.
Sorry for delay, since a large number of versions pending release in our process. RRN31.Q1-20-52-3 will be released first. Under normal circumstances, it is probably within1-2 weeks. Thank you!
Yuting.Chen

Hasaber8, 2021-05-06:
> @Hasaber8 ，thank.
> Sorry for delay, since a large number of versions pending release in our process. RRN31.Q1-20-52-3 will be released first. Under normal circumstances, it is probably within1-2 weeks. Thank you!
> Yuting.Chen

@chenyt9 first of all thank you, please do release the device trees too

chenyt9, 2021-05-12:
Hi,@all,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRN31.Q1-20-52-3
Thanks.

closed 2021-05-12 by chenyt9

SyberHexen, 2021-05-12:
@chenyt9 
Appreciate the push of this tag but it appears the kernel-devicetree was missed:
https://github.com/MotorolaMobilityLLC/kernel-devicetree/tags

electimon, 2021-06-01:
@chenyt9 bump

chenyt9, 2021-06-01:
@SyberHexen @electimon ,Thanks for the feedback, I will re-trigger the supplementary related files.

SyberHexen, 2021-06-18:
> @SyberHexen @electimon ,Thanks for the feedback, I will re-trigger the supplementary related files.

Any update on this?
Thanks in advance.

electimon, 2021-07-05:
@chenyt9 @mbissaromoto Please, release DTS sources for MMI-RRN31.Q1-20-52-3

ilia3367, 2021-07-08:
Hi! Can you please release kernel sources (Kernel, techpacks, DeviceTree) for Moto G100/Edge S? Thank you!

electimon, 2021-07-24:
@chenyt9 :(

ilia3367, 2021-07-24:
> @chenyt9 :(

I get the impression that these people just give standard, meaningless answers...

chenyt9, 2021-09-22:
For the tag RRN31.Q1-20-52-3， device tree is here：
https://github.com/MotorolaMobilityLLC/kernel-display-devicetree/releases/tag/MMI-RRN31.Q1-20-52-3
https://github.com/MotorolaMobilityLLC/kernel-camera-devicetree/releases/tag/MMI-RRN31.Q1-20-52-3
https://github.com/MotorolaMobilityLLC/kernel-devicetree/releases/tag/MMI-RRN31.Q1-20-52-3