MotorolaMobilityLLC/kernel-msm #340: [REQUEST]: Kernel release for RPF31.Q1-21-20 - Defender aka Def

Hasaber8, 2021-04-08:
RPF31.Q1-21-20 was pushed a few days ago, would really appreciate the kernel source being released
Bump @awheeler-mmi @mbissaromoto

chenyt9, 2021-05-06:
@Hasaber8 ，
Sorry for delay，this version is already in our process and will be released as soon as possible. Under normal circumstances, it is probably within two weeks. Thank you!

Hasaber8, 2021-06-19:
Bump @chenyt9 
Any updates on this?

chenyt9, 2021-06-21:
@Hasaber8 ，Sorry, I am urging the approval process . Once approved, release the version as soon as possible.

chenyt9, 2021-06-25:
Version released：

https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RPF31.Q1-21-20

Thanks

closed 2021-06-25 by chenyt9

Hasaber8, 2021-06-25:
Thank you @chenyt9 
Please look into this :- https://github.com/MotorolaMobilityLLC/kernel-msm/issues/350