MotorolaMobilityLLC/kernel-msm #341: [REQUEST] Kernel Release for QPX30.30-Q3-38-69 - gaump

raza231198, 2021-04-09:
QPX30.30-Q3-38-69 has been published few days ago, kindly upload kernel source as soon you can.
Bump @awheeler-mmi

mbissaromoto, 2021-04-16:
Released: https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-QPX30.30-Q3-38-69

closed 2021-04-16 by mbissaromoto

Hasaber8, 2021-04-16:
@mbissaromoto you missed https://github.com/MotorolaMobilityLLC/kernel-devicetree

mbissaromoto, 2021-04-16:
Opps, my bad. Sorry!
Check again, please

Hasaber8, 2021-04-16:
> Opps, my bad. Sorry!
> Check again, please

thank you!

raza231198, 2021-04-16:
> > Opps, my bad. Sorry!
> > Check again, please
> 
> thank you!

Device Tree Released For Guamp 👀

Hasaber8, 2021-04-16:
> > > Opps, my bad. Sorry!
> > > Check again, please
> > 
> > 
> > thank you!
> 
> Device Tree Released For Guamp 👀

https://github.com/MotorolaMobilityLLC/kernel-devicetree/blob/android-10-release-qpx30.30-q3-38-69/qcom/bengal-guamp-common-overlay.dtsi
they are there