MotorolaMobilityLLC/kernel-msm #343: [Request] Motorola G 30 (caprip) Kernel Sources (RRCS31.Q1-3-34-1-2)

Venkat3620, 2021-04-14:
please publish  kernel ,dts,techpack drivers  for Motorola G 30 (caprip)

chenyt9, 2021-05-06:
@Venkat3620 
Sorry for miss this msg，this version is already in our process and will be released as soon as possible. Under normal circumstances, it is probably within
two weeks. Thank you!

Venkat3620, 2021-05-06:
Thanks for response

GarfieldFernando, 2021-05-28:
@chenyt9 any update about for caprip?

Kuntitled, 2021-05-30:
Hoping this comes out soon

Venkat3620, 2021-05-30:
> @chenyt9 any update about caprip?

Source is already out but there are no device tree files.

https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-RRB31.Q1-3-34-3

Kuntitled, 2021-05-30:
> > @chenyt9 any update about caprip?
> 
> Source is already out but there are no device tree files.
> 
> https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-RRB31.Q1-3-34-3

Isn't that for capri specifically?

Venkat3620, 2021-05-30:
> > > @chenyt9 any update about caprip?
> > 
> > 
> > Source is already out but there are no device tree files.
> > https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-RRB31.Q1-3-34-3
> 
> Isn't that for capri specifically?

there is another tag MMI-RRB31.Q1-3-34-1 too that exactly has same code as this

chenyt9, 2021-05-31:
Hi， @Venkat3620 
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRCS31.Q1-3-34-1-2
Thanks for your patient！

closed 2021-05-31 by chenyt9

chenyt9, 2021-06-01:
@Venkat3620 ,
Thanks for the feedback, I will check the device tree files and confirm the related files.