MotorolaMobilityLLC/kernel-msm #344: [Request] Moto G10 - RRB31.Q1-3-34-3 kernel sources

rustycl0ck, 2021-04-16:
Can you please publish kernel sources for this device. Thanks!

rustycl0ck, 2021-05-05:
@motorolamobility any updates or ETA?

rustycl0ck, 2021-05-06:
bump @mbissaromoto

chenyt9, 2021-05-06:
@rustycl0ck ，Sorry for the delay, this version is already in our process and will be released as soon as possible. Under normal circumstances, it is probably within two weeks. Thank you!

rustycl0ck, 2021-05-06:
Thanks a lot for the info @chenyt9. Good to know that it is in pipeline. No worries about the delay. Prioritize safety and well-being of the staff first. Thanks!

chenyt9, 2021-05-13:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRB31.Q1-3-34-3

closed 2021-05-13 by chenyt9

Hasaber8, 2021-05-13:
@chenyt9 kernel-device-trees repo wasn't pushed . Please do so for caprip and nio

rustycl0ck, 2021-05-14:
Thanks a lot for the kernel sources @chenyt9. Just waiting for the device tree now.