MotorolaMobilityLLC/kernel-msm #346: [REQUEST]: Where is my camera and display dtsi? > QPX30.30-Q3-38-69

kytoaq, 2021-04-19:
### Where is dtsi for display and camera?

They have recently pushed the <b>MMI-QPX30.30-Q3-38-69</b> TAG for the [SM4250 Platform (Bengal)](https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPX30.30-Q3-38-69)
When trying to compile the kernel source manually I found that some files are missing

kernel-devicetree: qcom / camera
kernel-devicetree: qcom / display
kernel-devicetree: qcom / bindings / display / qcom

Where are the camera dtsi? Here I share a log about what is lost:
[bengal-camera-sensor-mot-guamp-overlay.dtsi](https://del.dog/whereismymissingdtsi.txt)

Could you please include it? In the log of the kernel-devicetree repository have set a [.gitignore](https://github.com/MotorolaMobilityLLC/kernel-devicetree/commit/fb501d9ae53db9f820859531c2f6bc47c29ddc23)

As additional information for their opensource Team, someone OEM's have shared those files for kernel 4.19 (bengal / lito / kona)

[OnePlus](https://github.com/OnePlusOSS/android_kernel_oneplus_sm4250/tree/oneplus/SM4250_Q_10.0/arch/arm64/boot/dts/vendor/qcom)
[Oppo](https://github.com/oppo-source/kernel_msm-4.19/tree/find-x2-r/arch/arm64/boot/dts/vendor)
[Xiaomi](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/lime-q-oss/arch/arm64/boot/dts/vendor/qcom)

Please share it, it doesn't matter if you have to write it to a new repository. Thanks

Don't forget to include the other files for the other models:
capri
cebu
borneo

kytoaq, 2021-04-19:
I have found what is missing. However only the branches for Lito appear, please push the QPX branch

Excuse me for the inconvenience

Hasaber8, 2021-04-19:
https://github.com/MotorolaMobilityLLC/kernel-camera-devicetree/blob/MMI-QPX30.30-Q3-38-69/bengal-camera-sensor-mot-guamp-overlay.dtsi

Hasaber8, 2021-04-19:
https://github.com/MotorolaMobilityLLC/kernel-display-devicetree/tree/MMI-QPX30.30-Q3-38-69

closed 2021-04-19 by kytoaq