MotorolaMobilityLLC/kernel-msm #348: [REQUEST](Moto G60/G40 fusion)(Hanoip/Hanoi) RRI31.Q1-42-24-4 Kernel source release

Hasaber8, 2021-05-02:
Please release the Kernel source as soon as possible.
- Thanks, Rohan.
- bump @awheeler-mmi @mbissaromoto

chenyt9, 2021-05-06:
Hi, @Hasaber8 ,this version is already in our process and will be released as soon as possible. Under normal circumstances, it is probably within two weeks. Thank you!

Hasaber8, 2021-05-06:
Great work today @chenyt9 , Thank you for all the updates!

Hasaber8, 2021-05-27:
> Hi, @Hasaber8 ,this version is already in our process and will be released as soon as possible. Under normal circumstances, it is probably within two weeks. Thank you!

Any updates @chenyt9 ?

chenyt9, 2021-05-28:
@Hasaber8 ，
Version released：https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRI31.Q1-42-24-4
Thanks for your patient.

closed 2021-05-28 by chenyt9

Hasaber8, 2021-06-27:
@chenyt9 Seems like with RRI31.Q1-42-42-3 a touch lag fix was shipped, can you please release this version?
thank you