MotorolaMobilityLLC/kernel-msm #349: GPL | Source request for Motorola Moto Edge (racer) (RPD31.Q4U-39-26-4)

SyberHexen, 2021-05-11:
Kindly requesting source for the following branches for this tag:
- kernel
- kernel-devicetree
- techpack-camera
- techpack-display
- techpack-video
- vendor-qcom-opensource-audio-kernel

Thanks in advance!
~Andrew

SyberHexen, 2021-05-19:
bump @awheeler-mmi @mbissaromoto @chenyt9

SyberHexen, 2021-07-28:
@chenyt9 sorry to bump again, any idea if this request has made any progress yet?

SyberHexen, 2021-08-18:
Well this is just ridiculous...
I'll be seeking legal action over the poor display on this device plus  the violation of releasing open source in timely matter...

 I feel like I've been patient, polite and been ignored over the months while I've watched countless edge users displays go out..

Totally users did thier part but you guys too busy flooding the market with devices rather than working on the shortfalls of current issues...

mbissaromoto, 2022-03-08:
Hi all,

It is released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RPD31.Q4U-39-26-4

Att,
Marcelo

SyberHexen, 2022-03-09:
> Hi all,
> 
> It is released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RPD31.Q4U-39-26-4
> 
> Att, Marcelo

Thank you!

closed 2022-03-09 by SyberHexen