MotorolaMobilityLLC/kernel-msm #350: GPL | Source Request Moto G(50) (Ibiza) XT2137-1_IBIZA_RRF31.Q1-59-33

Hasaber8, 2021-05-13:
Kindly requesting source for the following repos for this tag:-
kernel-msm
kernel-devicetree

Hasaber8, 2021-05-13:
Bump @awheeler-mmi @mbissaromoto @chenyt9

PeterCxy, 2021-05-18:
Also including the following repos for the kernel modules:

-  kernel-msm-5.4-techpack-{display,video,camera}
- vendor-qcom-opensource-*
- motorola-kernel-modules

chenyt9, 2021-06-01:
@Hasaber8 @PeterCxy ，
This version is already in our process and will be released as soon as possible. Under normal circumstances, it is probably within two weeks. Thank you!

Hasaber8, 2021-06-19:
Bump @chenyt9 
Any updates on this?

chenyt9, 2021-06-28:
Version released:

https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRF31.Q1-59-33

Thanks.

closed 2021-06-28 by chenyt9

Hasaber8, 2021-06-28:
@chenyt9 techpack and device trees aren't pushed. Please check and release them please

mbissaromoto, 2022-05-04:
Hi @Hasaber8  @decaduto

Missing repos are now released. Please, let me known if something else is missing.