MotorolaMobilityLLC/kernel-msm #351: [REQUEST] Moto G Power 2021 (Borneo) QZBS30.Q4-43-43-2 sources

iprouteth0, 2021-05-25:
Please release sources for moto g power 2021 codename Borneo
Build QZBS30.Q4-43-43-2.  Thank you

chenyt9, 2021-06-01:
@iprouteth0 ，
This version is already in our process and will be released as soon as possible. Under normal circumstances, it is probably within two weeks. Thank you!

iprouteth0, 2021-06-21:
Still waiting on this.  I do not yet see it in the releases tags.

thrymgjol, 2021-07-03:
When will these sources be released?

iprouteth0, 2021-07-08:
This is starting to get rediculous.  It has been a month and a half since I submitted this request.  If these are not released soon, I will begin prices of reporting Motorola for GPL violations to FSF (license-violation@gnu.org)

chenyt9, 2021-07-09:
This version is already in our process, we will release both QZB30.Q4-43-130 and QZBS30.Q4-43-43-2 first. May release in these two days.

iprouteth0, 2021-07-09:
Thank you for the update

chenyt9, 2021-07-09:
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QZBS30.Q4-43-43-2
Thanks for your patient

closed 2021-07-09 by chenyt9