MotorolaMobilityLLC/kernel-msm #352: [Request] Moto G10 - RRB31.Q1-3-48-1 kernel sources and Device Tree

rustycl0ck, 2021-06-01:
Previous kernel sources were shared in #344 and #338. Many thanks for that.

But the **Device Tree** is missing in both of them. Please share the device tree as well as the newer kernel sources.

chenyt9, 2021-07-12:
Hi，@rustycl0ck 
sorry for miss this issue. This version is already in our process and will be released as soon as possible. Under normal circumstances, itusually takes two weeks. In addition, Device Tree related releases will also be arranged. Thank you!

chenyt9, 2021-07-26:
@rustycl0ck .
version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRB31.Q1-3-48-2
Thanks.

closed 2021-07-26 by chenyt9

tom1033f, 2021-09-02:
@chenyt9 @rustycl0ck 
Apologies if I'm being dumb but I'm hoping you can clarify for me - rustycl0ck says:
"Previous kernel sources were shared in #344 and #338. Many thanks for that.
But the Device Tree is missing in both of them. Please share the device tree as well as the newer kernel sources."

Then a link is posted to MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRB31.Q1-3-48-2
... but it has the same name format, the same content and is the same size as #344 and #338.

And the contents don't look like what I'd expect for a full device tree.

Apologies if I missed something but could you please advise - is there a device tree for the G10? Where can I get it? Again, sorry if I'm being dumb.
Tom

rustycl0ck, 2021-09-02:
Hi @tom1033f , Yes, it's not intuitive, but the device tree is in a different repo - https://github.com/MotorolaMobilityLLC/kernel-devicetree/releases/tag/MMI-RRB31.Q1-3-48-2

Although the required files were shared, my zero knowledge of android prevented me from verifying anything or making use of these source files. I hope better for you. 🙂

tom1033f, 2021-09-03:
> Hi @tom1033f , Yes, it's not intuitive, but the device tree is in a different repo - https://github.com/MotorolaMobilityLLC/kernel-devicetree/releases/tag/MMI-RRB31.Q1-3-48-2
>
> Although the required [...]

Thanks @rustycl0ck !!!