MotorolaMobilityLLC/kernel-msm #356: [Question] Branch: marshmallow-6.0.1-release-smelt, Help changing the maximum sample rate for the accelerometer in the kernel.

lightkeima, 2021-07-07:
I'm trying to change the maximum sample rate for the accelerometer in marshmallow-6.0.1-release-smelt. 
The sample rate currently has maximum value is 50Hz when I ran "dumpsys senserservice".
I've read the source code and found a m4sensorhub driver which containing m4sensorhub-accelerometer driver.
However, I've never worked on a device with sensorhub before, I don't know how it must be done in the right way.
Can you please tell me how do I do that?

lightkeima, 2021-11-02:
Yeah I'm still trying to search for the solution

asbraga-motorola, 2022-08-03:
Hi @ntvthuyen,
This forum is primarily for requesting open source code publication or questions related to it.
Please use the Moto Community Forum for all other questions:
https://forums.lenovo.com/t5/Motorola-Community/ct-p/MotorolaCommunity

closed 2022-08-03 by asbraga-motorola