MotorolaMobilityLLC/kernel-msm #357: [REQUEST] Motorola Moto G30 (Caprip) (RRCS31.Q1-3-34-1-4) [GPL]

GarfieldFernando, 2021-07-10:
Hi, everyone!
Could u release **kernel, full techpack and kernel-devicetree sources** for this tag **RRCS31.Q1-3-34-1-4**? Please!

Thank you! =)
 ~ Fernando

ferhzzoma, 2021-07-11:
bump @chenyt9

chenyt9, 2021-07-12:
@ferhzzoma ，this version is already in our process and will be released as soon as possible. Under normal circumstances, itusually takes two weeks. Thank you!

chenyt9, 2021-07-23:
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRCS31.Q1-3-34-1-4
Thanks

closed 2021-07-23 by chenyt9

mikeNG, 2021-09-26:
@chenyt9 It looks like you forgot to push `MMI-RRCS31.Q1-3-34-1-4` for https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-data-kernel , can you please take a look?