MotorolaMobilityLLC/kernel-msm #358: [REQUEST] Kernel Release for RPX31.Q2-58-16 - guamp

BotReza, 2021-07-11:
RPX31.Q2-58-16 has been published few days ago, kindly upload kernel source as soon you can.

BotReza, 2021-07-11:
@mbissaromoto @awheeler-mmi

leole0leoleo, 2021-07-24:
when will it be posted?

leole0leoleo, 2021-07-25:
@chenyt9 @mbissaromoto @awheeler-mmi

leole0leoleo, 2021-08-18:
@mbissaromoto pls

mbissaromoto, 2021-09-06:
Hi @leole0leoleo 
Really sorry for the delay. It is now released.

closed 2021-09-06 by mbissaromoto

RazaDroid, 2021-09-15:
@mbissaromoto https://github.com/MotorolaMobilityLLC/kernel-msm-4.19-techpack-display techpack is not pushed

mbissaromoto, 2021-09-18:
@RazaDroid  as I replied on the other repo (and just to keep it in track here): it should be fine now.

Please, note that the tag was applied on top of wrong commits before, for some repos. It should be fine now, so please fetch the update. And sorry about that :(