MotorolaMobilityLLC/kernel-msm #360: [Request] Moto G9 Power (cebu) - RZC31.Q2-57-12 kernel sources

mistersmee, 2021-07-12:
Kindly requesting source for the following repos for this tag "RZC31.Q2-57-12"

    kernel-msm
    kernel-devicetree
    kernel-camera-devicetree
    kernel-display-devicetree
    kernel-msm-4.19-techpack-camera
    kernel-msm-4.19-techpack-display
    kernel-msm-4.19-techpack-video
    vendor-qcom-opensource-audio-kernel
    vendor-qcom-opensource-wlan-qcacld-3.0
    vendor-qcom-opensource-wlan-fw-api
    vendor-qcom-opensource-wlan-qca-wifi-host-cmn

Thanks in advance!

mistersmee, 2021-07-12:
Bump @awheeler-mmi @chenyt9 @mbissaromoto

mistersmee, 2021-07-22:
Bump @chenyt9

mistersmee, 2021-07-28:
Bump @chenyt9

mistersmee, 2021-08-07:
Bump @chenyt9 @awheeler-mmi @mbissaromoto

mistersmee, 2021-08-31:
If there were any changes between the newer tags, having those would be great as well.

mistersmee, 2021-08-31:
Bump @chenyt9 @awheeler-mmi @mbissaromoto 
It's been 1.5 months, any updates as to when it's going to be released?

mistersmee, 2021-09-13:
Bump @chenyt9 @mbissaromoto @awheeler-mmi 

With the recent releases for guamp and nio, I hoped that this would be released too. It's been 2 months since I opened the issue.

mistersmee, 2021-09-29:
@chenyt9 @mbissaromoto @awheeler-mmi 

The second security update since the update to Android 11 has even dropped, and still kernel source for cebu have not been released.
It will almost be 3 months since opening the issue soon. When will cebu sources be released?

mistersmee, 2021-11-05:
Bump @chenyt9 @mbissaromoto @motorolamobility @vinimed 

It has been just about 4 months since I opened the issue, when are Cebu sources going to be released? With the recent releases for a few devices, I had hoped that cebu would be one of them.

vinimed, 2022-03-17:
Hello, all!
Version Released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RZCS31.Q2-57-12-1

Note: We had already started the process when you updated the request from 12-1 suffix to a superior. If you still need the informed releases above, please consider to open a new issue.

Thank you for your patience and comprehension.

closed 2022-03-17 by vinimed