MotorolaMobilityLLC/kernel-msm #362: [REQUEST] Moto G Power 2021 (Borneo) - MMI-QZBS30.Q4-43-43-2 - all device tree/dts sources

iprouteth0, 2021-07-15:
Kindly release the following sources (all device tree/dts) for borneo(moto g power 2021).  Tag MMI-QZBS30.Q4-43-43-2 

kernel-devicetree
kernel-camera-devicetree
kernel-display-devicetree

Other sources have already been released.
Thanks
James

ferhzzoma, 2021-07-16:
bump @chenyt9 pls

chenyt9, 2021-07-26:
@ferhzzoma @iprouteth0 ,pls refer to 'MMI-QZBS30.Q4-43-43-2' tag on devicetree repo.
like:
https://github.com/MotorolaMobilityLLC/kernel-devicetree/releases/tag/MMI-QZBS30.Q4-43-43-2
https://github.com/MotorolaMobilityLLC/kernel-display-devicetree/releases/tag/MMI-QZBS30.Q4-43-43-2
https://github.com/MotorolaMobilityLLC/kernel-camera-devicetree/releases/tag/MMI-QZBS30.Q4-43-43-2

closed 2021-07-26 by chenyt9