MotorolaMobilityLLC/kernel-msm #365: [REQUEST] Moto g20 (java) RTA31.42-5-5 sources

TheMalachite, 2021-07-30:
Hello, could you release the kernel and kernel modules source for the Motorola g20 (java) from RTA31.42-5-5 please ?

That's a Unisoc Tiger T700 powered device however

TheMalachite, 2021-07-30:
@awheeler-mmi @chenyt9 @mbissaromoto

TheMalachite, 2021-09-22:
Bump @chenyt9

DavidDeath21, 2022-01-16:
kernel source error... please share Source

Tokoteke, 2022-01-16:
could you release the kernel and kernel modules source for the Motorola g20 (java) from RTA31.42-5-5 please ?

Zonik94, 2022-01-17:
Please take our request into consideration, a new kernel update has just come out if possible to source this thousands of users and devs will be grateful @awheeler-mmi @chenyt9 @mbissaromoto 🙏

zmoky2468, 2022-01-19:
Hello, could you release the kernel and kernel modules source for the Motorola g20 (java) from RTA31.42-5-5 please ?

That's a Unisoc Tiger T700 powered device however
Hail UwU.

Claravallac, 2023-02-07:
Hello moto???
Source??? 

@chenyt9

Claravallac, 2023-02-26:
@chenyt9
@awheeler-mmi
@TinnoOSS
@asbraga-motorola
@mbissaromoto

Bump it, please

closed 2024-05-09 by asbraga-motorola