MotorolaMobilityLLC/kernel-msm #367: [REQUEST] Moto Edge S - RRN31.Q2-12-14-7 - all device tree/dts sources

gyfx, 2021-08-09:
Kindly release the following sources (all device tree/dts) for nio(moto edge s). Tag  RRN31.Q2-12-14-7

kernel-devicetree
kernel-camera-devicetree
kernel-display-devicetree

Thanks

gyfx, 2021-08-16:
bump @chenyt9  pls

chenyt9, 2021-08-19:
@gyfx ，this version is already in our process and will be released as soon as possible. Under normal circumstances, itusually takes two weeks. Thank you!

SGCMarkus, 2021-08-28:
I suppose those sources will then work fine too for the G100 (nio, RRTS31.Q1-84-21-1-2), as the devices are basically the same, and just marketed for different regions?

mukul2259, 2021-08-30:
Bump, would love to see sources so that developers can work on it.

chenyt9, 2021-09-02:
@gyfx ,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRN31.Q2-12-14-7
Thanks

Hasaber8, 2021-09-02:
@chenyt9 again, device tree weren't pushed 😥

electimon, 2021-09-02:
@chenyt9 since may we've been waiting for these sources...

https://github.com/MotorolaMobilityLLC/kernel-msm/issues/335#issuecomment-839707469

Hasaber8, 2021-09-02:
@awheeler-mmi you promised us proper kernel release with device trees, but again and again kernels are getting released without them ಠ︵ಠ

Hasaber8, 2021-09-02:
Also consistently false timelines are given, making the whole matter worse, seems like you guys haven't recovered from https://github.com/charleseb leaving the team

SGCMarkus, 2021-09-03:
yep, kinda a shame, these are such nice devices
I dont get why you wouldnt release the kernel devicetrees, other manufacturers manage to do that too

chenyt9, 2021-09-06:
@SGCMarkus ，@Hasaber8 ，
Sorry for the missing device-tree code, I have re-triggered a new release process. Will release as soon as possible. Thanks for your patience.

chenyt9, 2021-09-06:
@Hasaber8 ,I'm sorry I don't quite understand what you mean of the "false timelines", what problem are you referring to? The code is released as is, there should be no errors.

Hasaber8, 2021-09-06:
> @Hasaber8 ,I'm sorry I don't quite understand what you mean of the "false timelines", what problem are you referring to? The code is released as is, there should be no errors.

I was referring to, when you say in under normal circumstances it will release in 2 weeks but it takes 4-5 weeks. I understand nothing is normal of the current circumstances.

Also please add device trees to the new process, the delay just forces us to prebuild the kernel.
Also please look into https://github.com/MotorolaMobilityLLC/kernel-msm/issues/323
Its been pending for 7 months now. Again forcing us to prebuild the kernel.

Hasaber8, 2021-09-06:
Also @chenyt9 please don't take it in the wrong way, the kernel release has been very slow lately compared to last few years.

chenyt9, 2021-09-10:
@SGCMarkus ，@Hasaber8 ， 
For kernel device tree, pls refer to:
https://github.com/MotorolaMobilityLLC/kernel-display-devicetree/releases/tag/MMI-RRN31.Q2-12-14-7
https://github.com/MotorolaMobilityLLC/kernel-camera-devicetree/releases/tag/MMI-RRN31.Q2-12-14-7
https://github.com/MotorolaMobilityLLC/kernel-devicetree/releases/tag/MMI-RRN31.Q2-12-14-7

Hasaber8, 2021-09-10:
Thank you @chenyt9

SGCMarkus, 2021-09-10:
Thanks @chenyt9