MotorolaMobilityLLC/kernel-msm #371: Please upload kernel sources for new lineup

J0SH1X, 2021-08-30:
hey , i would like to rqeuest the upload of the kernel sources of the motorola edge s edge 20 and edge 20 pro.

it would be pretty nice if you could publish them as GPL does require it.

thx in advance

chenyt9, 2021-08-31:
Hi,@J0SH1X ，
Here is version released for edge s，can it meet you need？
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRN31.Q1-20-52-3
for other product， could you please specify the version you need. Thank.

Hasaber8, 2021-08-31:
> Hi,@J0SH1X ，
> Here is version released for edge s，can it meet you need？
> https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRN31.Q1-20-52-3
> for other product， could you please specify the version you need. Thank.

You guys forgot to push device trees for this release, please do that

chenyt9, 2021-09-22:
@J0SH1X ， Pls use the device tree code first, I will check the code and repost the tag for MMI-RRN31.Q1-20-52-3 .
https://github.com/MotorolaMobilityLLC/kernel-display-devicetree/releases/tag/MMI-RRN31.Q2-12-14-7
https://github.com/MotorolaMobilityLLC/kernel-camera-devicetree/releases/tag/MMI-RRN31.Q2-12-14-7
https://github.com/MotorolaMobilityLLC/kernel-devicetree/releases/tag/MMI-RRN31.Q2-12-14-7
Thanks

chenyt9, 2021-09-22:
Hi， @J0SH1X ，device trees is here
https://github.com/MotorolaMobilityLLC/kernel-display-devicetree/releases/tag/MMI-RRN31.Q1-20-52-3
https://github.com/MotorolaMobilityLLC/kernel-camera-devicetree/releases/tag/MMI-RRN31.Q1-20-52-3
https://github.com/MotorolaMobilityLLC/kernel-devicetree/releases/tag/MMI-RRN31.Q1-20-52-3

closed 2021-09-22 by chenyt9