MotorolaMobilityLLC/kernel-msm #373: [Request] Moto One Fusion (astro) QPLS30.62-41-8 Techpack/Device Tree (Kernel 4.9)

PHLA7, 2021-09-12:
Hello, Motorola maintainers.
I am requesting the device tree for the Motorola one fusion (astro).
Including: 
kernel-msm-4.9-techpack-video
kernel-msm-4.9-techpack-camera
kernel-msm-4.9-techpack-display

chenyt9, 2021-09-22:
Hi, @PHLA7 
The release process of the version has been triggered and it may be need take about two weeks. We will release it as soon as possible. But there is no found separate device tree git for kernel-msm-4.9.
The released versions are:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPLS30.62-23-2-3
I don’t know if we can meet your needs first. please wait patiently, thanks!

chenyt9, 2021-11-02:
Hi, @PHLA7 ,
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QPLS30.62-41-8
Thanks.

closed 2021-11-02 by chenyt9