MotorolaMobilityLLC/kernel-msm #374: GPL | Motorola Moto G Power (2021) Borneo QZBS30.Q4-43-120-5 Source release

MrNerd, 2021-09-21:
Can you please release the kernel source for this phone?

Thank you

 Motorola Moto G Power (2021) Borneo QZBS30.Q4-43-120-5

MrNerd, 2021-09-21:
I think i have the Device tree. If you want i can send it to you and you can tell me if its correct.

MrNerd, 2021-09-21:
I have no clue sir.. Im just as new as you. I found a tree generator script made in python and that's how I got it.

thebestofall007, 2021-09-21:
> I have no clue sir.. Im just as new as you. I found a tree generator script made in python and that's how I got it.

That's OK. I'll still take the tree if you could send it please. If I don't find out if it's the right one myself, perhaps someone else can if I upload it to XDA Forums

MrNerd, 2021-09-22:
https://github.com/MrNerd/MotoGPower_2020_DT

chenyt9, 2021-09-22:
Hi，@MrNerd 
The release process of the version has been triggered and it may be need take about two weeks. We will release it as soon as possible，includes the device tree code. 
The released versions are:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QZB30.Q4-43-130
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QZBS30.Q4-43-43-2
I don’t know if we can meet your needs first. please wait patiently, thanks!

MrNerd, 2021-09-23:
Thank you for the reply. Can you also tell me how to compile for Borneo because i dont see a Defconfig file for that phone

arkansawdave74, 2021-10-08:
What's up guys. I just got this this phone too and i might be of some help. I'm not great at building roms, but I have several lolipop and several nougat builds on afh. Sometimes I get lucky. Would you be willing to share your tree with me as well, @MrNerd ?

MrNerd, 2021-10-08:
Sure its on my Github

chenyt9, 2021-11-01:
@MrNerd ,
For the config , pls refer to:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-QZBS30.Q4-43-120-5.txt
Version released：
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-QZBS30.Q4-43-120-5
Thanks.

closed 2021-11-01 by chenyt9