MotorolaMobilityLLC/kernel-msm #375: Edge 2021 RRM31.Q3-23-29-5 kernel source

Dabug123, 2021-09-22:
Would love to see this available along with a tree for development

chenyt9, 2021-09-22:
Hi, @Dabug123 
The release process of the version has been triggered and it may be need about two weeks. We will release it as soon as possible. Please wait patiently, thanks!

Dabug123, 2021-09-22:
Thanks.There wasnt source on the previous ver or the device release ..

Thanks for your time

Dabug123, 2021-09-29:
I hope we are working towards a release of a source soon following this repo it seems multiple devices are very behind ..Great device overall gonna be sad if development is hindered by lack of source

chenyt9, 2021-11-03:
@Dabug123
Sorry for delay, Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRM31.Q3-23-29-5
Thanks.

closed 2021-11-03 by chenyt9