MotorolaMobilityLLC/kernel-msm #376: [Request] Motorola Moto G30 (caprip) GPL sources (RRCS31.Q1-3-56-2)

mikeNG, 2021-09-29:
Could you please release kernel sources for caprip 'RRCS31.Q1-3-56-2' build?

Includes the following repositories:

```
kernel-msm
kernel-devicetree
kernel-camera-devicetree
kernel-display-devicetree
kernel-msm-4.19-techpack-camera
kernel-msm-4.19-techpack-display
kernel-msm-4.19-techpack-video
vendor-qcom-opensource-audio-kernel
vendor-qcom-opensource-data-kernel
vendor-qcom-opensource-wlan-fw-api
vendor-qcom-opensource-wlan-qca-wifi-host-cmn
vendor-qcom-opensource-wlan-qcacld-3.0
motorola-kernel-modules
```

ferhzzoma, 2021-10-10:
bump @chenyt9 @awheeler-mmi

chenyt9, 2021-11-03:
@ferhzzoma ,@Hasaber8 ,@mikeNG,
Version released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRCS31.Q1-3-56-2
Thanks.

closed 2021-11-03 by chenyt9

mikeNG, 2021-11-03:
> @ferhzzoma ,@Hasaber8 ,@mikeNG, Version released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-RRCS31.Q1-3-56-2 Thanks.

@chenyt9 just like https://github.com/MotorolaMobilityLLC/kernel-msm/issues/357#issuecomment-927304645 it seems you have not pushed the tag for `vendor-qcom-opensource-data-kernel`. Can you please upload it?