MotorolaMobilityLLC/kernel-msm #385: [REQUEST]: Android 12 - sofiap_sprout SPR32.54 Kernel Sources

kytoaq, 2021-11-05:
Since Android 11 couldn't send the Android 11 kernel update (see #353), I am forced to ask for this tag. I'd appreciate it if you could upload it.

Excuse my desperation for requesting it, but quite a few releases of different devices have passed and it is a bit annoying to learn that this device cannot be released. We have been trying to get the source of this kernel for more than 8 months but the answer is negative. Please release the version as soon as possible, since Android 12 should not be a problem.

Thanks you :)

mbissaromoto, 2022-08-01:
Hi @kytoaq  

After you opened this request here, I've published the tag that was requested on #353. I will close this issue here, ok?
 
If you need a request for this or any other tag for Android 12, please update this issue accordingly.

Regards,
Marcelo

kytoaq, 2022-08-01:
Hi!.

thanks for answering this issue. I would really appreciate if you could upload this Tag for the Trinket platform.

S0PRS32.44-11-10-2-10

Thanks you.

asbraga-motorola, 2022-08-17:
Hi @kytoaq 
The version is released: https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-S0PRS32.44-11-10-2-10

closed 2022-08-17 by asbraga-motorola