MotorolaMobilityLLC/kernel-msm #387: [Request] Moto Razr 5G (smith) RPS31.Q1-40-34-15-10-2

webgeek1234, 2021-11-20:
Please release kernel source, kernel device trees, and any relevant kernel module and techpack source.

mbissaromoto, 2022-04-13:
Hi @webgeek1234 
It is now released: https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-RPS31.Q1-40-34-15-10

Att,
Marcelo / Vinicius

closed 2022-04-13 by mbissaromoto

Artem7shenin, 2023-11-26:
> Hi @webgeek1234 It is now released: https://github.com/MotorolaMobilityLLC/kernel-msm/tree/MMI-RPS31.Q1-40-34-15-10
> 
> Att, Marcelo / Vinicius

I know that I’m writing off topic, but it seems to me that this is the last place where they can help me.  fold tell me where the file is located in which there is a value that sets the shutter speed value for other applications, now it is 1/4 but I would like to increase this, for example to 500 seconds or more, how can I do this

Smartphone: moto edge 20, android 12, global rom