MotorolaMobilityLLC/kernel-msm #388: Moto G Play (2021) [guamna] has decoding problems with 10-bit HEVC

Pentaphon, 2021-12-12:
Steps: 
1) Download https://jell.yfish.us/media/jellyfish-10-mbps-hd-hevc-10bit.mkv
2) Play it on https://play.google.com/store/apps/details?id=com.brouken.player&hl=en_US&gl=US
3) Won't play, see screenshot below for error you get. 

[image]

nsherbina1999, 2021-12-17:
Are you sure it's not a SW decode problem?

Pentaphon, 2021-12-17:
> Are you sure it's not a SW decode problem?

I'm not sure. I've been told by other video player devs on github that this device has buggy decoders. I'd love to help debug this with somebody and get a permanent fix for this.

asbraga-motorola, 2022-08-03:
Hi @Pentaphon,
This forum is primarily for requesting open source code publication or questions related to it.
Please use the Moto Community Forum for all other questions:
https://forums.lenovo.com/t5/Motorola-Community/ct-p/MotorolaCommunity

closed 2022-08-03 by asbraga-motorola

Pentaphon, 2022-08-03:
@asbraga-motorola can you please take my bug report to the people at Motorola who take bug reports? I don't need community support. I need a firmware bug fixed.