MotorolaMobilityLLC/kernel-msm #393: [Request] GPL | Motorola g31 Source release (Device Tree, Kernal code and vendor etc.)

harshzalavadiya, 2022-02-03:


vinimed, 2022-03-09:
Hello @harshzalavadiya ,

Can you please specify the release number you want? By having the release number we can focus on releasing the exact version you need.

Vinicius

harshzalavadiya, 2022-03-10:
@vinimed thanks, it's RRWB31.Q3-46-64-2-2

chenyt9, 2022-03-25:
@harshzalavadiya ，
this version is already in our process and will be released as soon as possible. Under normal circumstances, itusually takes two weeks. Thank you!

chenyt9, 2022-04-22:
@harshzalavadiya ,
Verison released:
https://github.com/MotorolaMobilityLLC/kernel-mtk/releases/tag/MMI-RRWB31.Q3-46-64-2-2

closed 2022-04-22 by chenyt9

VTVISHWANATH, 2022-10-12:
Can we get the latest version? 

Also does it include Kernel + device tree + vendor tree/blobs or only kernel tree?

harshzalavadiya, 2022-10-12:
@VTVISHWANATH I think it will be better to create new issue

VTVISHWANATH, 2022-10-26:
@motorolamobility please upload source for below
GPL | Source request for Motorola Moto G31[retin] (XT2173-2) RRWBS31.Q3-46-85-7
https://github.com/MotorolaMobilityLLC/kernel-msm/issues/450